ianfixes/nmea_plus #2: "argument out of range" error in eta method

sternschnuppe, 2016-09-22:
## If ETA is not set (i.e. "0-0 24:60" or "0-0 00:00") the method stops throwing "argument out of range (ArgumentError)".

- Error message:

/home/user/.rvm/gems/ruby-2.3.0/gems/nmea_plus-1.0.12/lib/nmea_plus/message/ais/vdm_payload/vdm_msg5.rb:26:in `initialize': argument out of range (ArgumentError)

from /home/user/.rvm/gems/ruby-2.3.0/gems/nmea_plus-1.0.12/lib/nmea_plus/message/ais/vdm_payload/vdm_msg5.rb:26:in `new'

from /home/user/.rvm/gems/ruby-2.3.0/gems/nmea_plus-1.0.12/lib/nmea_plus/message/ais/vdm_payload/vdm_msg5.rb:26:in `eta'

from Decode_NMEA.rb:72:in `block in main'

from /home/user/.rvm/gems/ruby-2.3.0/gems/nmea_plus-1.0.12/lib/nmea_plus.rb:103:in `block in each_complete_message'

from /home/user/.rvm/gems/ruby-2.3.0/gems/nmea_plus-1.0.12/lib/nmea_plus.rb:56:in `block in each_message'

from /home/user/.rvm/gems/ruby-2.3.0/gems/nmea_plus-1.0.12/lib/nmea_plus.rb:46:in `each_line'

from /home/user/.rvm/gems/ruby-2.3.0/gems/nmea_plus-1.0.12/lib/nmea_plus.rb:46:in `each_message'

from /home/user/.rvm/gems/ruby-2.3.0/gems/nmea_plus-1.0.12/lib/nmea_plus.rb:81:in `each_complete_message'

from Decode_NMEA.rb:17:in `main'

- Example sentence: !AIVDM,2,1,6,B,53JNnF42BiCWTP7?3KE<B18uHE:222222222220l1h>5540Ht<P000000000,0*6A
!AIVDM,2,2,6,B,00000000000,2*21
- Workaround:
`begin eta = message.ais.eta; rescue eta = nil; end`

ifreecarve, 2016-09-22:
> unfortunately grave accents are interpreted so I can't use any of these sentences

I'm not sure what you mean.  Can you paste those sentences here in a [fenced code block](https://help.github.com/articles/creating-and-highlighting-code-blocks/#fenced-code-blocks)?

ifreecarve, 2016-09-22:
In these examples, it looks like the ETA is meant to be nil: zeros for the day and month, 24 for hour and 60 for the seconds.

sternschnuppe, 2016-09-22:
For the sake of completeness I added a sentence with a backtick.

closed 2016-09-23 by ifreecarve

ifreecarve, 2016-09-23:
Thanks, I used your data in rspec tests

PR #4 (2016-09-23): date/times: make better assumptions about year and recognize nil
Fixes #2 and #3