ianfixes/nmea_plus #3: ETA doesn't handle year rollover

ifreecarve, 2016-09-22:
Code [here](https://github.com/ifreecarve/nmea_plus/blob/98eccf80cbd996356feb1d25762be3e1595293a6/lib/nmea_plus/message/ais/vdm_payload/vdm_msg5.rb#L25-L27) and probably other places

Our ETA function chooses the current year.  It should choose next year if the ETA month is lower than the month in the local timezone.

sternschnuppe, 2016-09-22:
Thanks! 
It would be also fantastic if you could add a method for ETA raw data ("12-31 23:59")!

ifreecarve, 2016-09-23:
With AIS, there is no "raw" data in the way that there would be for an NMEA string.  The integer values for each component of the date (month, day, hour, minute) are pulled out of fixed-length bit fields after the message has been 6b decoded.  I'm returning the date object because it's a very appropriate data structure to hold all these fields.  The year is implied to be the current (and/or upcoming) year, and the seconds are implied to be 00, so that's what I return.  

TL;DR, if you want it in string format, you'll have to build the string yourself because I never get it in string format to begin with.

ifreecarve, 2016-09-23:
Fixed in 1.0.13

closed 2016-09-23 by ifreecarve