ianfixes/nmea_plus #14: Add support for AIS type 17

ifreecarve, 2016-10-04:
Need example data

alamut, 2019-01-02:
not sure if this gem is still maintained, but here's some sample data for type 17:
```
AIVDM,1,1,,A,A02VdoPCa@lB@7iTD<tP102LwCl=wn@3?Jbb,0
AIVDM,1,1,,B,A02VdoPCa@lB@7iTD>@P102KwCl=wnD3?Jbb,0
AIVDM,1,1,,A,A02VdoPCa@lB@7iTDApP102IwCl=wnL3?Jbb,0
AIVDM,1,1,,B,A02VdoPCa@lB@7iTDH0P102GwCl=wnT3?Jbb,0
AIVDM,1,1,,A,A02VdoPCa@lB@7iTDKPP102EwCl=wnd3?Jbb,0
```

ianfixes, 2019-01-03:
That's excellent, thank you!  It's still maintained, although I've done as much development as I can without access to more specs or example data.  So this is a big help!

Have you found the gem itself to be useful?

ianfixes, 2019-01-03:
Extracting the payload field as a hex-encoded string, I notice that it only changes in a few spots:
```
7c6450cf2004009cfd3d0dff64033daaaa
7c6450e42004009bfd3d0dff65033daaaa
7c64511e20040099fd3d0dff67033daaaa
7c64518020040097fd3d0dff69033daaaa
7c6451b820040095fd3d0dff6b033daaaa
     ^^^       ^         ^
```

This suggests that there are fields within this payload, which some documentation suggests might be encoded as RTCM2.  Do you have any knowledge about that?

alamut, 2019-01-03:
no idea, i was using messages with type 1, as far as i'm interested only in ship's coords.
but here're some messages decoded with a python package https://github.com/schwehr/libais
```
!AIVDM,1,1,,B,A02VdoPCa@lB@7iTD>@P102KwCl=wnD3?Jbb,0*01
{'id': 17, 'repeat_indicator': 0, 'mmsi': 2731230, 'spare': 0, 'x': 33.526666666666664, 'y': 44.61666666666667, 'spare2': 0}

!AIVDM,1,1,,A,A02VdoPCa@lB@7iTDApP102IwCl=wnL3?Jbb,0*47
{'id': 17, 'repeat_indicator': 0, 'mmsi': 2731230, 'spare': 0, 'x': 33.526666666666664, 'y': 44.61666666666667, 'spare2': 0}

!AIVDM,1,1,,B,A02VdoPCa@lB@7iTDH0P102GwCl=wnT3?Jbb,0*1B
{'id': 17, 'repeat_indicator': 0, 'mmsi': 2731230, 'spare': 0, 'x': 33.526666666666664, 'y': 44.61666666666667, 'spare2': 0}
```

PR #116 (2019-01-04): Add AIS message type 17
Fixes #14 

Thanks to @alamut for providing sample data
closed 2019-01-04 by ianfixes