ianfixes/nmea_plus #22: AIVDO message causes uninitialized constant error

jpcavanagh, 2017-01-08:
When processing this AIVDO message an error occurs:

!AIVDO,1,1,,,B3P9=>h007vPUK7Amc@03wlUsP06,0*49

Couldn't instantiate a NMEAPlus::Message::AIS::VDO object: uninitialized constant NMEAPlus::Message::AIS::VDO
Did you mean?  NMEAPlus::Message::AIS::VDM
["/var/lib/gems/2.3.0/gems/nmea_plus-1.0.17/lib/nmea_plus/message_factory.rb:61:in `rescue in dynamically_get_message_object'", "/var/lib/gems/2.3.0/gems/nmea_plus-1.0.17/lib/nmea_plus/message_factory.rb:59:in `dynamically_get_message_object'", "/var/lib/gems/2.3.0/gems/nmea_plus-1.0.17/lib/nmea_plus/message_factory.rb:77:in `create'", "parser.y:10:in `_reduce_2'", "/var/lib/gems/2.3.0/gems/racc-1.4.14/lib/racc/parser.rb:271:in `_racc_do_parse_c'", "/var/lib/gems/2.3.0/gems/racc-1.4.14/lib/racc/parser.rb:271:in `do_parse'", "/var/lib/gems/2.3.0/gems/nmea_plus-1.0.17/lib/nmea_plus/generated_parser/tokenizer.rb:30:in `scan_str'", "/var/lib/gems/2.3.0/gems/nmea_plus-1.0.17/lib/nmea_plus/generated_parser/tokenizer.rb:86:in `parse'", "nmea_server.rb:22:in `handle_client'", "nmea_server.rb:53:in `block in <main>'"]

ifreecarve, 2017-01-09:
Thanks!  I'll check it out

ifreecarve, 2017-01-09:
Fixed in 1.0.18

closed 2017-01-09 by ifreecarve

jpcavanagh, 2017-01-11:
> Closed #22 <https://github.com/ifreecarve/nmea_plus/issues/22> via #23

Thanks for fixing.  Unfortunately I see a similar error with $IIVWT,
 $IIVWM and $AITXT

PR #23 (2017-01-09): AIVDO support
fixes #22