ianfixes/nmea_plus #24: $IIVWT causes error with unitialized constant

jpcavanagh, 2017-01-11:
$IIVWT,058.0,R,15.9,N,8.2,M,29.4,K*4A

Couldn't instantiate a NMEAPlus::Message::NMEA::IIVWT object: uninitialized constant NMEAPlus::Message::NMEA::IIVWT
["/var/lib/gems/2.3.0/gems/nmea_plus-1.0.19/lib/nmea_plus/message_factory.rb:64:in `rescue in dynamically_get_message_object'", "/var/lib/gems/2.3.0/gems/nmea_plus-1.0.19/lib/nmea_plus/message_factory.rb:62:in `dynamically_get_message_object'", "/var/lib/gems/2.3.0/gems/nmea_plus-1.0.19/lib/nmea_plus/message_factory.rb:80:in `create'", "parser.y:9:in `_reduce_1'", "/var/lib/gems/2.3.0/gems/racc-1.4.14/lib/racc/parser.rb:271:in `_racc_do_parse_c'", "/var/lib/gems/2.3.0/gems/racc-1.4.14/lib/racc/parser.rb:271:in `do_parse'", "/var/lib/gems/2.3.0/gems/nmea_plus-1.0.19/lib/nmea_plus/generated_parser/tokenizer.rb:30:in `scan_str'", "/var/lib/gems/2.3.0/gems/nmea_plus-1.0.19/lib/nmea_plus/generated_parser/tokenizer.rb:86:in `parse'", "nmea_server.rb:21:in `handle_client'", "nmea_server.rb:67:in `block in <main>'"]

ifreecarve, 2017-01-11:
Unfortunately, this error is by design -- I throw exceptions for message types that are not defined.  

Do you have any information on this message type and the meaning of each field?

jpcavanagh, 2017-01-12:
IIVWT comes from a Raymarine ST60 "integrated instrument" system which uses an internal protocol between devices and it put onto the NMEA bus via a dedicated Raymarine instrument box.  Some of sentences used are very old and can be found here http://www.nmea.org/Assets/100108_nmea_0183_sentences_not_recommended_for_new_designs.pdf

In my test files I also found the HVM (magnetic heading) which is in the document referenced above
$IIHVM,2,W*36
Couldn't instantiate a NMEAPlus::Message::NMEA::IIHVM object: uninitialized constant NMEAPlus::Message::NMEA::IIHVM

Lastly I noticed that AITXT caused an error - I believe this comes from a Comar AIS transponder.
$AITXT,01,01,70,Entered a DSC Receivable Window Chan A*46
Couldn't instantiate a NMEAPlus::Message::NMEA::AITXT object: uninitialized constant NMEAPlus::Message::NMEA::AITXT

I understand that any unknown message or un-decodable string throws an error;   let me know if you want to know about further valid messages which are un-implemented.

John

ifreecarve, 2017-01-20:
I'm making progress on that document, which you can track here: #28 

Would you feel comfortable contributing?

jpcavanagh, 2017-01-21:
Sounds like good progress.  What contribution would be helpful?   I can
create a file from a live NMEA stream which uses some of these sentences if
that would help testing ?

ifreecarve, 2017-01-22:
I won't say no to having real-life examples of these sentences.  At the moment I'm just inventing them based on what the specs look like, and hoping that they're accurate.  

I'm also wondering if you'd feel comfortable copying the style that I'm using to implement these messages, and submitting your own: https://github.com/ifreecarve/nmea_plus/pull/28/commits/1916d5822b765263c6f238b14beec05eebde142a

ifreecarve, 2017-02-01:
This is now fixed in the branch:
https://github.com/ifreecarve/nmea_plus/tree/2017-01-20_nmea_not_recommended

I'll be continuing to work on the document (only 17 message types to go now) before I merge and release a new gem.

closed 2017-02-02 by ifreecarve

ifreecarve, 2017-02-02:
Should be all set in https://rubygems.org/gems/nmea_plus/versions/1.0.20

PR #28 (2017-02-02): Add NMEA "not recommended" messages
* Lots of documentation added to fields (although for just the deprecated messages)
* Fixes #25 
* Fixes #24 
* Fixes #27