ianfixes/nmea_plus #120: MID 311 Falsely decoding to Bermuda instead of Bahamas

aSipiere, 2021-03-17:
https://github.com/ianfixes/nmea_plus/blob/39ff69dd24806315ecc2b40ca6ca68e2c8459e12/lib/nmea_plus/message/ais/vdm_payload/mmsi_info.rb#L182

Spotted this one by chance recently where one system was using third party data and another was decoding with NMEA Plus.

https://www.itu.int/en/ITU-R/terrestrial/fmd/Pages/mid.aspx has Bahamas as `308`, `309` & `311`, so the fix should be as simple as `when 311 then 44` on ln 182.

ianfixes, 2021-03-18:
Hi @aSipiere, thanks for contributing this!  This ITU page is exactly what I was looking for when I wrote this library, but the google search terms were just too general for me to get what I was looking for.  

If you don't mind my asking, how do you use this gem and how has it served your needs?

aSipiere, 2021-03-18:
No worries mate happy to help. Glad to see you're still semi-active on here. I'm not a ruby dev but I work on a project where we use this to decode a live AIS stream for a live map, sorry I can't be more specific. afaik we have the decoder wrapped to add some nmea v4 features, but otherwise it's been great, this is the first and only problem I think we've spotted in 2 years.

ianfixes, 2025-04-05:
This is fixed and released in version `1.1.1`.  Sorry for the delay in such a trivial fix, it had to get in behind a pretty big cleanup.

closed 2025-04-05 by ianfixes